uoip/g2opy #1: Symbol not found: __ZN3g2o10HyperGraph12UnassignedIdE

ubuntuslave, 2018-02-23:
Compilation and Installation succeeds. However, when trying `>>> import g2o`, I get the following error:

```
ImportError: dlopen(/usr/local/lib/python3.6/site-packages/g2o.cpython-36m-darwin.so, 2): Symbol not found: __ZN3g2o10HyperGraph12UnassignedIdE
  Referenced from: /usr/local/lib/python3.6/site-packages/g2o.cpython-36m-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/python3.6/site-packages/g2o.cpython-36m-darwin.so
```

I'm using Python 3.6 from Homebrew. In addition, I had compiled/installed `g2o` from source.

SpitfireX, 2018-05-27:
[image]

Geohot found a workaround for this [on stream](https://www.twitch.tv/tomcr00s3).

kenkyusha, 2018-06-01:
To make it clear the file which needed to be changed is following:
g2o/python/core/hyper_graph.h
on line 71 : 
"id"_a=HyperGraph::UnassignedId)
replaced by
"id"_a=-1) 
and line 82:
"id"_a=-2 //HyperGraph::InvalidId)
by 
"id"_a=-2 )
after that 
make -j8

xiaojingli, 2018-06-29:
Sorry, use the latest vision which has changed hyper_graph.h, but still meet the same problem...

CChape, 2018-07-24:
Sorry too, but getting the same problem and hyper_graph.h has been changed..

Apo-, 2018-11-02:
I'm not familiar with pybind so I'm not sure if this a library issue[**1**], but here's a slightly more detailed description and my assessment of the issue.

`UnassignedId` and `InvalidId` are integer `static const` class members of `HyperGraph` and `static const` members of integral type are a weird beast in C++land, because they can be initialised inside the class definition and be used in certain contexts (integral constant expressions), but in order to be odr-used they still need to be defined. This essentially means that they can be used in constant expressions without being defined and the compiler will replace their values (so no symbols will be generated) and the linker will never know those variables ever existed.

Concretely, in this case these two variables are not defined so a symbol is not generated for them, but apparently a symbol is required (haven't dug into why that's the case in macOs).

One solution is to define them so that the symbols are generated:
```c++
// .cpp file
namespace g2o {
    const int HyperGraph::UnassignedId;
    const int HyperGraph::InvalidId;
} // namespace g2o
```
Doing that produces a library where when we query its symbols we get:
```
$ nm -U /.../lib/python3.6/site-packages/g2o.cpython-36m-darwin.so | grep "UnassignedId\|InvalidId"
00000000002d91e0 S __ZN3g2o10HyperGraph12UnassignedIdE
00000000002d91e4 S __ZN3g2o10HyperGraph9InvalidIdE
```
Of course this way you might run into odr violations if the author of g2o decides to define these variables.

The other solution is to make it so that the symbols are not required, by using temporaries:
```diff
-  "id"_a = HyperGraph::InvalidId)
+  "id"_a = static_cast<decltype(HyperGraph::InvalidId)>(HyperGraph::InvalidId))
```
```diff
- "id"_a = HyperGraph::UnassignedId)
+ "id"_a = static_cast<decltype(HyperGraph::UnassignedId)>(HyperGraph::UnassignedId))
```

[**1**]: On linux the symbol is not defined either in the shared library, but the problem doesn't seem to occur. This could be because of how the symbols are loaded on linux and macos (so the problem is there, but it's not hit until later(?)) or it could be a pybind issue
On Linux:
```
$ nm -D /.../lib/g2o.cpython-36m-x86_64-linux-gnu.so | grep "UnassignedId\|InvalidId"
$ 
```

PRABHUGUPTA, 2019-01-09:
still getting the same error in ubuntu18.04 with python3.7
plss help............
error is as follows

python
Python 3.7.1 (default, Dec 14 2018, 19:28:38) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import g2o
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /home/royal_duke/anaconda3/lib/python3.7/site-packages/g2o.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN3g2o10HyperGraph12UnassignedIdE