uoip/g2opy #4: Problem when making

cyc790, 2018-04-22:
**when executing the" make" instructor, it turns out unused parameter warning like the following:**

[ 99%] Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
In file included from /home/cyc/g2o/build/g2opy/python/g2o.cpp:19:0:
/home/cyc/g2o/build/g2opy/python/core/base_unary_edge.h:33:40: warning: unused parameter ‘m’ [-Wunused-parameter]
 void declareBaseUnaryEdge(py::module & m) {
                                        ^
In file included from /home/cyc/g2o/build/g2opy/python/g2o.cpp:20:0:
/home/cyc/g2o/build/g2opy/python/core/base_binary_edge.h:40:41: warning: unused parameter ‘m’ [-Wunused-parameter]
 void declareBaseBinaryEdge(py::module & m) {
                                         ^
In file included from /home/cyc/g2o/build/g2opy/python/g2o.cpp:25:0:
/home/cyc/g2o/build/g2opy/python/core/linear_solver.h:38:39: warning: unused parameter ‘m’ [-Wunused-parameter]
 void declareLinearSolver(py::module & m) {
                                       ^
**and then it does not move forward!**

**how to solve this?  Appreciated for any suggestions!**

thomasfermi, 2018-05-29:
same problem here

closed 2018-06-02 by cyc790

1280032578, 2022-07-13:
I also encountered the same problem.
Is there a solution? thanks.