uoip/g2opy #5: g2o.cpp build error

DellBrother, 2018-04-24:
[ 99%] Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
In file included from /home/dell/Downloads/g2opy/python/g2o.cpp:12:0:
/home/dell/Downloads/g2opy/python/core/eigen_types.h: In function ‘void g2o::declareEigenTypes(pybind11::module&)’:
/home/dell/Downloads/g2opy/python/core/eigen_types.h:185:82: error: no matches converting function ‘x’ to type ‘double (class Eigen::Quaternion<double>::*)() const’
         .def("x", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::x)
                                                                                  ^
In file included from /usr/local/include/eigen3/Eigen/Geometry:42:0,
                 from /home/dell/Downloads/g2opy/g2o/core/eigen_types.h:31,
                 from /home/dell/Downloads/g2opy/build/g2o/config.h:26,
                 from /home/dell/Downloads/g2opy/g2o/core/g2o_core_api.h:13,
                 from /home/dell/Downloads/g2opy/g2o/core/hyper_graph.h:40,
                 from /home/dell/Downloads/g2opy/python/core/hyper_graph.h:5,
                 from /home/dell/Downloads/g2opy/python/g2o.cpp:3:
/usr/local/include/eigen3/Eigen/src/Geometry/Quaternion.h:75:52: note: candidates are: Eigen::QuaternionBase<Derived>::NonConstCoeffReturnType Eigen::QuaternionBase<Derived>::x() [with Derived = Eigen::Quaternion<double>; Eigen::QuaternionBase<Derived>::NonConstCoeffReturnType = double&]
   EIGEN_DEVICE_FUNC inline NonConstCoeffReturnType x() { return this->derived().coeffs().x(); }
                                                    ^
/usr/local/include/eigen3/Eigen/src/Geometry/Quaternion.h:66:44: note:                 Eigen::QuaternionBase<Derived>::CoeffReturnType Eigen::QuaternionBase<Derived>::x() const [with Derived = Eigen::Quaternion<double>; Eigen::QuaternionBase<Derived>::CoeffReturnType = const double&]
   EIGEN_DEVICE_FUNC inline CoeffReturnType x() const { return this->derived().coeffs().coeff(0); }

DellBrother, 2018-05-04:
I apologize!
I install Eigen 3.3.4, pass build!

thanks,

closed 2018-05-04 by DellBrother

pineappleKID, 2019-04-17:
eigen-3.3.5 fail
eigen-3.3.4 pass