uoip/g2opy #7: undefined symbol: cs_di_schol

thomasfermi, 2018-05-29:
Installation completed, but when I import g2o I get 

ImportError: /mypath/python3.6/site-packages/g2o.cpython-36m-x86_64-linux-gnu.so: undefined symbol: cs_di_schol

Im using Linux Mint 18.3 (which is based on Ubuntu 16.04)

cyc790, 2018-06-02:
I encountered the same problem,have you solved it? @thomasfermi

thomasfermi, 2018-06-02:
Hi @cyc790 
No, sorry. I gave up.

ferreram, 2018-07-04:
Hi @thomasfermi and @cyc790,
I had the same issue and I solved it by installing libqglviewer-dev.

Hope it can work for you too.
Cheers!

district10, 2018-10-21:
> Hi @thomasfermi and @cyc790,
> I had the same issue and I solved it by installing libqglviewer-dev.
> 
> Hope it can work for you too.
> Cheers!

reinstalled my libqglviewer-dev and it works now, thanks a lot

AmaurX, 2018-11-20:
Hi!
I tried to install everything with anaconda (and python3.7). The install works as well, but I get the same problem : 
```ImportError: /mypath/python3.7/site-packages/g2o.cpython-36m-x86_64-linux-gnu.so: undefined symbol: cs_di_schol```

Notice that `mypath` here is actually "HOME/anaconda3/envs/py37/lib/". 

I tried reinstalling  libqglviewer-dev but it didn't change anything. It might be because the .so file is in our custom path in anaconda and not in /usr/lib

Then I reconstructed everything outside of anaconda (with python3.5) and it works.

Any idea on how to make things work in anaconda?

lvgeng, 2019-06-09:
> Hi @thomasfermi and @cyc790,
> I had the same issue and I solved it by installing libqglviewer-dev.
> 
> Hope it can work for you too.
> Cheers!

I am not so sure if it is a recent change.
Currently, there are two versions of  libqglviewer-dev.

libqglviewer-dev-qt4 and libqglviewer-dev-qt5

The key is libqglviewer-dev-qt4 need to be installed. Well... kind of hard to understand but it it works...
And I assume that qt4 or qt5 only plays a part in visualization, I guess it would not affect the performance of graph optimization.

Hope so...

pro100kot, 2019-06-18:
Hi!
For me worked:
1. install libqglviewer-dev (or libqglviewer-dev-qt4 like @lvgeng noticed)
2. install libsuitesparse-dev
3. **rebuild g2opy**

Maybe it will be useful to someone.

a1qq2345609, 2019-09-03:
Hi!
I encountered the problem of Segmentation fault when I directly import g2o at the very beginning of python 3.5.2:
`>>> import g2o`
`Segmentation fault (core dumped)`
But when I import g2o after importing a module that doesn't exist, undefined symbol error occurs:
`>>> import ggg`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`ImportError: No module named 'ggg'`
`>>> import g2o`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`ImportError: /usr/lib/python3/dist-packages/g2o.cpython-35m-x86_64-linux-gnu.so: undefined symbol: cs_di_schol`
I've tried to install libqglviewer-dev (or libqglviewer-dev-qt4) and libsuitesparse-dev and rebuild g2opy mentioned by @pro100kot but nothing worked.
I don't know whether these two problems are the same. Either do I know how to fix them...

marctuscher, 2020-02-02:
Hi,
installing http://libqglviewer.com/download.html from source worked for me.

shaoxh, 2021-02-14:
> Hi,
> installing http://libqglviewer.com/download.html from source worked for me.
Hi, 
tried this, but still not work.
then I remove libqglviewer-dev-qt5 and  install libqglviewer-dev-qt4, not work neither. @CydKnocking 
I have a question. 
I have qmake4 and qmake5 both in my environment. does it influence qglviewer thing?
#50 @marctuscher

cacadam89, 2021-12-09:
I am running into this same issue now, and the previously suggested fixes are not working. I am on Ubuntu 20.04.3 LTS, and when I try to install `libqglviewer`, the only options are `libqglviewer2-qt5`, `libqglviewer-dev-qt5`, `libqglviewer-doc`, and `libqglviewer-headers`. I have tried them all and none of this fixes my issues. I also tried @marctuscher's link (http://libqglviewer.com/download.html) to install from source and that didnt fix it either.

If this is only related to the visualization part, is there an easy way to change things so that I can still import g2o and use the optimization functionality w/o errors?

If anyone has any other suggestions, I would love to hear them!

freesynapse, 2022-01-21:
Have the same problem. libqglviewer-dev and libsuitsparse-dev installed, build without errors. Import fails with "/usr/lib/python3/dist-packages/g2o.cpython-38-x86_64-linux-gnu.so: undefined symbol: cs_di_schol"

randomlylelo, 2022-01-29:
I had this same problem, solved by deleting g2opy folder, re-cloning it, but then do the change on #48 before building it. Afterwards, it just worked.

Strohhut1, 2022-05-11:
My solution was the same as from @randomlylelo:

System: Ubuntu 20.04, Eigen3, python3.8

1. remove g2o folder
2. remove `g2o.cpython-...` file from /usr/lib/python3/dist-packages
3. clone g2o again
4. change `g2opy/python/core/eigen_types.h` according to #48 or #16 (I don't now which is better #16 worked for me)
5. install `libqglviewer` from source (http://libqglviewer.com/download.html) and `libqglviewer-dev-qt5` with apt-get
6. follow build instructions