uoip/g2opy #11: g2opy do not compile on mac os mojave

toter, 2018-10-10:
Hi, I'm trying to compile g2opy on mac os mojave but i'm getting some errors. Any help would be appreciated. Please let me know if any more information is necessary.

Python versions installed: 2.7.15 and 3.7.0

### cmake output:
--------------

-- Compiling on OSX
-- Could NOT find CHOLMOD (missing: CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES) 
-- Building LGPL code as static library (affects license of the binary)
-- Compiling with OpenGL support
CMake Warning at cmake_modules/FindQGLViewer.cmake:1 (FIND_PACKAGE):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:148 (FIND_PACKAGE)

Qt5 not found. Install it and set Qt5_DIR accordingly
-- Could NOT find QGLVIEWER (missing: QGLVIEWER_INCLUDE_DIR QGLVIEWER_LIBRARY) 
-- Compiling with Clang
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CHOLMOD_INCLUDE_DIR
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python
   used as include directory in directory /Users/toter/Downloads/twitchslam/tmp/g2opy/python

-- Configuring incomplete, errors occurred!
See also "/Users/toter/Downloads/twitchslam/tmp/g2opy/build/CMakeFiles/CMakeOutput.log".

### CMakeOutput.log:
--------------

The system is: Darwin - 18.0.0 - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /Library/Developer/CommandLineTools/usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is AppleClang, found in "/Users/toter/Downloads/twitchslam/build/g2opy/python/CMakeFiles/3.12.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /Library/Developer/CommandLineTools/usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"
[... 494 lines cut ...]

Determining if the function cheev_ exists passed with the following output:
Change Dir: /Users/toter/Downloads/twitchslam/build/g2opy/python/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_33211/fast"
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/cmTC_33211.dir/build.make CMakeFiles/cmTC_33211.dir/build
Building C object CMakeFiles/cmTC_33211.dir/CheckFunctionExists.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -DLAPACK_USE_F2C  -DCHECK_FUNCTION_EXISTS=cheev_ -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk   -o CMakeFiles/cmTC_33211.dir/CheckFunctionExists.c.o   -c /usr/local/Cellar/cmake/3.12.3/share/cmake/Modules/CheckFunctionExists.c
Linking C executable cmTC_33211
/usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_33211.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=cheev_ -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names   CMakeFiles/cmTC_33211.dir/CheckFunctionExists.c.o  -o cmTC_33211 -framework Accelerate -framework Accelerate 

Performing C++ SOURCE FILE Test HAS_FLTO succeeded with the following output:
Change Dir: /Users/toter/Downloads/twitchslam/build/g2opy/python/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_93002/fast"
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/cmTC_93002.dir/build.make CMakeFiles/cmTC_93002.dir/build
Building CXX object CMakeFiles/cmTC_93002.dir/src.cxx.o
/Library/Developer/CommandLineTools/usr/bin/c++    -Wall -fPIC -DHAS_FLTO -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk   -flto -o CMakeFiles/cmTC_93002.dir/src.cxx.o -c /Users/toter/Downloads/twitchslam/build/g2opy/python/CMakeFiles/CMakeTmp/src.cxx
Linking CXX executable cmTC_93002
/usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_93002.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++    -Wall -fPIC -DHAS_FLTO -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names   CMakeFiles/cmTC_93002.dir/src.cxx.o  -o cmTC_93002 -flto 

Source file was:
int main() { return 0; }

tresf, 2019-01-09:
> Could not find a package configuration file provided by "Qt5" with any of
> the following names:
> 
> ```
> Qt5Config.cmake
> qt5-config.cmake
> ```

Using homebrew (for example), you tell `cmake` where  Qt5 is with the following command:

```bash
export CMAKE_PREFIX_PATH=$(brew --prefix qt)
```

its-not-a-bug-its-a-feature, 2019-06-09:
Try install suitesparse - this resolved the error for me

brew install suitesparse

metobom, 2022-04-14:
> Try install suitesparse - this resolved the error for me
> 
> brew install suitesparse

Solved for me too on Kubuntu 20.04. 

`sudo apt-get install -y libsuitesparse-dev`