uoip/g2opy #15: Only private modules found

varun19299, 2018-10-31:
Installed on `python 3.6`, anaconda. Built g2o successfully. 

Ran : `export PYTHON_PATH=PATH-to-g2o/g2o:$PYTHONPATH` .

Able to import g2o, but only private modules found. Unable to use `g2o.SparseOptimizer`.

```
Python 3.5.2 (default, Nov 23 2017, 16:37:01)
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import g2o
>>> g2o.__
g2o.__class__(         g2o.__dir__(           g2o.__format__(        g2o.__gt__(            g2o.__le__(            g2o.__name__           g2o.__package__        g2o.__reduce_ex__(     g2o.__sizeof__(        g2o.__subclasshook__(
g2o.__delattr__(       g2o.__doc__            g2o.__ge__(            g2o.__hash__(          g2o.__loader__         g2o.__ne__(            g2o.__path__           g2o.__repr__(          g2o.__spec__
g2o.__dict__           g2o.__eq__(            g2o.__getattribute__(  g2o.__init__(          g2o.__lt__(            g2o.__new__(           g2o.__reduce__(        g2o.__setattr__(       g2o.__str__(
```

Haotian-Zhang, 2018-12-26:
The same question as yours, did you find any solutions?

varun19299, 2018-12-31:
Installing with sudo (I was okay with that since I was using it in a container) seemed to work.

Naturally that would be in the system python. I don't quite understand the issue though? Do some modules under the Cmake file have permission requirements?

EnriqueSolarte, 2019-02-18:
hi everybody,
@Haotian-Zhang. did you solve this issue? I am still on error.
```
AttributeError: module 'g2o' has no attribute 'BlockSolverSE3'
```

EnriqueSolarte, 2019-02-18:
I solved my problem. 
I was trying to install on python 3.6... it doesn't work. I installed on python 3.5.2
I installed in the default python3 of UBUNTU 16.04.
```
python3 setup.py install
``` 
I also created a conda env. 
```
conda create -n g2o_env python=3.5.2
conda activate g2o_env 
conda python setup.py install
```
Remember that setup.py script is under the directory project....

codelovin, 2019-04-17:
Have the solution been found without using the conda env?

varun19299, 2019-04-17:
Yeah but that isn't a recommended way

medakk, 2019-04-24:
For some reason, g2opy is creating a python3.7 library. If I create a virtualenv with python3.7, I am able to use it. The generated so file is called "g2o.cpython-37m-x86_64-linux-gnu.so". How do I force g2opy to compile with 3.6 or any arbitrary version?

(Also, renaming g2o.cpython-37m-x86_64-linux-gnu.so  to g2o.cpython-36m-x86_64-linux-gnu.so gives an import error for the version mismatch)

kaixinbear, 2020-03-03:
I install g2opy with sudo ,but it doesn't work either...

minhkhang1795, 2020-03-05:
For those having the same problem, first don't run any `export PYTHON_PATH=PATH-to-g2o/g2o:$PYTHONPATH` because the `g2o` folder is written in C++ and not what you want. 

As discussed [here](https://github.com/uoip/g2opy/issues/9), you need to build for the python version you're using. Remove the `build` folder and re-run cmake with:
```
cmake -DPYBIND11_PYTHON_VERSION=3.6 ..
```

yiakwy-enterprise-roborock, 2020-04-15:
@minhkhang1795  @varun19299 @EnriqueSolarte 
This is due to the fact the External/pybind11/CMakeLists.txt computed wrong version of python.

After some modification you can get correction python library with correct extension name:

```txt
 yiakwy@yiakwy  ~/WorkSpace/Github/g2opy/build   master ● 
 cmake ..
-- Compiling on Unix
-- Found CHOLMOD and its dependencies
-- Building LGPL code as static library (affects license of the binary)
-- Compiling with OpenGL support
-- Could NOT find QGLVIEWER (missing: QGLVIEWER_INCLUDE_DIR QGLVIEWER_LIBRARY)
-- Compiling with GCC
-- pybind11 PYTHON_INCLUDE_DIR: /home/yiakwy/anaconda3/envs/py36/include/python3.6m
-- pybind11 PYTHON_LIBRARY: /home/yiakwy/anaconda3/envs/py36/lib/libpython3.6m.so.1.0
-- PYTHON_MODULE_PREFIX: /home/yiakwy/anaconda3/envs/py36
-- PYTHON_MODULE_EXTENSION: .cpython-36m-x86_64-linux-gnu.so
-- pybind11 v2.2.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yiakwy/WorkSpace/Github/g2opy/build

```