uoip/g2opy #18: cmake error

zqnnn, 2018-11-29:
[image]

grumpydevelop, 2018-12-16:
install suitesparse

meh369, 2019-02-14:
I'm getting the exact same error, I have installed Suitesparse but don't know what to do with it? Should I be setting some system variables? I'm using Windows 10

grumpydevelop, 2019-02-14:
@meh369 message me in telegram @osnriwugy, will see what i can to do

meh369, 2019-02-15:
Ahh, I resolved it by uninstalling the latest version to an older version and it worked! However now im getting make error : make[2]: *** [python/CMakeFiles/g2o.dir/g2o.cpp.obj] Error 1
make[1]: *** [python/CMakeFiles/g2o.dir/all] Error 2
make: *** [all] Error 2