uoip/g2opy #19: build fails on Jetson TX1

jmgl0003, 2019-01-22:
Hello I am  trying  to build it and I meet this error:

[ 99%] Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
In file included from /home/nvidia/Documents/g2opy-master/python/g2o.cpp:19:0:
/home/nvidia/Documents/g2opy-master/python/core/base_unary_edge.h:33:40: warning: unused parameter ‘m’ [-Wunused-parameter]
 void declareBaseUnaryEdge(py::module & m) {
                                        ^
In file included from /home/nvidia/Documents/g2opy-master/python/g2o.cpp:20:0:
/home/nvidia/Documents/g2opy-master/python/core/base_binary_edge.h:40:41: warning: unused parameter ‘m’ [-Wunused-parameter]
 void declareBaseBinaryEdge(py::module & m) {
                                         ^
In file included from /home/nvidia/Documents/g2opy-master/python/g2o.cpp:25:0:
/home/nvidia/Documents/g2opy-master/python/core/linear_solver.h:38:39: warning: unused parameter ‘m’ [-Wunused-parameter]
 void declareLinearSolver(py::module & m) {
                                       ^
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
python/CMakeFiles/g2o.dir/build.make:62: recipe for target 'python/CMakeFiles/g2o.dir/g2o.cpp.o' failed
make[2]: *** [python/CMakeFiles/g2o.dir/g2o.cpp.o] Error 4
CMakeFiles/Makefile2:1577: recipe for target 'python/CMakeFiles/g2o.dir/all' failed
make[1]: *** [python/CMakeFiles/g2o.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

doiken23, 2019-07-12:
I got same error in my macbook.
It seems to be the error because of the out of memory...

Keshav919, 2019-11-20:
change -j8 to -j4 and it should work

1280032578, 2022-07-13:
I use the make command instead of make -j8.
The program has been waiting at 99%.
Is there a solution? Thanks!!!