uoip/g2opy #22: undefined symbol

aivision2020, 2019-01-30:
ImportError: /home/user/Projects/sixdof/sixdof_env/lib/python3.6/site-packages/g2o.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN3g2o17csparse_extension14writeCs2OctaveEPKcPK12cs_di_sparseb
I'm using ubuntu 17.10

I don't know if this is related but I had trouble installing libqglviewer-dev:

Package libqglviewer-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libqglviewer-headers

I tried installing libqglviewer-headers, and also libqglviewr-dev-qt4. but to no avail.
Any help would be greatly appreciated. Thanks

aivision2020, 2019-01-31:
change to Ububutu 16.04.05. All good now

guohengkai, 2020-02-03:
Same error on Python 3.7 with Debian 9.

guohengkai, 2020-02-04:
Solved by disabling the CSparse extension building.