uoip/g2opy #23: Pose Graph Optimization

phildue, 2019-01-31:
Hi,
first of all thanks for providing the python bindings.
I'm trying to implement a Pose Graph Optimization based on some C++ code where they use the  edge: EdgeSE3ProjectXYZOnlyPose(). In this example they use the Xw property of this class to store the 3D locations of the points. These are required to computer the projection error.

In the python bindings I cannot access this property. Is this a bug? Also, I was wondering how you would use this class instead, or how to perform the pose graph optimization.

Thanks

phildue, 2019-02-04:
Follow up: If I just set the VertexSBAPoints as fixed, doesnt it perform pose graph optimization then?

medakk, 2019-05-29:
@phildue Did you figure out how to access the properties?