uoip/g2opy #27: Error while building g2o.cpp.obj

meh369, 2019-02-15:
[image]

meh369, 2019-02-15:
Im using window 10

meh369, 2019-03-04:
So I'm trying to compile the firmware, first I did Cmake g2o with suitesparse(Cholmod and its dependencies), Eigen3 and Qt5.4 and generated it to Visual Studio solution "g2o.sln" file(using VS-15 2017) that can build g2o. However after building INSTALL(shown in the snip) I get 3 errors in timeutil.h file and when I located the file it had a struct deftype statement but I couldnt figure out the actual error. Help? 
[image]

vhgnguyen, 2019-04-13:
hi @meh369  i am having the same error as you while building with VS-15-2017 x64. Did you solve this?

shrddr, 2019-04-20:
It has been fixed back in 2017, take a look here
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/stuff/timeutil.h
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/stuff/timeutil.cpp

meh369, 2020-01-21:
> hi @meh369 i am having the same error as you while building with VS-15-2017 x64. Did you solve this?

No, I switched to Linux OS...

closed 2020-01-21 by meh369