uoip/g2opy #29: Building on Windows [solution]

shrddr, 2019-04-22:
Here are the steps to build on Windows with MSVC 2017
* when specifying the generator in cmake-gui, pick x64 platform. you also need python-x64. 32bit target will lead to "compiler out of heap space"
* latest eigen release will not work, use 3.3.4 instead
* cholmod library
    * option 1 - build without it (limited functionality) - just comment out lines 4,17 in python\CMakeLists.txt and lines 10,17 in python\core\blocksolver.h
    * option 2 - build with vcpkg
        * `set VCPKG_DEFAULT_TRIPLET=x64-windows`
        * `vcpkg install suitesparse clapack openblas`
        * replace cmake_modules\FindBLAS, FindCholmod, FindCSparse, FindLAPACK, FindSuiteSparse with similar files from https://github.com/RainerKuemmerle/g2o
        * when picking a generator for cmake, instead of "use default native compilers" select "specify toolchain file" - vcpkg\scripts\buildsystems\vcpkg.cmake
        * untick BUILD_CSPARSE
        * Configure, make sure cmake prints "Found CHOLMOD and its dependencies", then Generate
    * option 3 - build manually - remember you will also need BLAS and LAPACK
* there's a problem with timeval/timezone structs in g2o\stuff\timeutil.h and timeutil.cpp. don't know the root cause but here is the dirty workaround
    * move timezone declaration to .h
    * change both typedefs to simple struct
    * rename timeval->timeval1, timezone->timezone1
* compilation of bindings takes up to 8Gb of memory so 64-bit compiler is a must. build with `"c:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\amd64\MSBuild.exe" g2o.sln /property:Configuration=Release`
* before running `python setup.py install`, in setup.py change `'./lib/g2o*.so'` to `'./bin/Release/g2o*.pyd'`. 
* if using BLAS and LAPACK, also copy these dll's from bin/Release to python\Lib\site-packages

vhgnguyen, 2019-05-05:
Thanks for the instruction. i gave up building it on Windows 2 weeks ago and now see your the instruction. However i still have same error with "timeutil". Can you please upload snipet of your code how your rewrite it?

shrddr, 2019-05-05:
Here you go https://github.com/shrddr/g2opy

vhgnguyen, 2019-05-05:
Work like a charm. Thank you very much 😄

Haotian-Zhang, 2019-05-06:
@shrddr Hi, thank you for your contributions to the windows version. Currently, I am using vcpkg to install every package. It only allows me to install eigen 3.3.7, which will cause the build to fail. Could you please tell me how to install eigen3.3.4? I'm new to windows and my environment is windows 10, VS 2015, and using anaconda as my python base. Thank you in advance!

vhgnguyen, 2019-05-06:
> @shrddr Hi, thank you for your contributions to the windows version. Currently, I am using vcpkg to install every package. It only allows me to install eigen 3.3.7, which will cause the build to [...]

The easiest way is that you can just download eigen 3.3.4 from 

`http://eigen.tuxfamily.org/index.php?title=News:Eigen_3.3.4_released!`

 and extract it. Then in cmake_modules\FindEigen3.cmake set EIGEN3_INCLUDE_DIR to that folder because eigen is header only so you dont need to build it

Haotian-Zhang, 2019-05-06:
> The easiest way is that you can just download eigen 3.3.4 from
>
> `http://eigen.tuxfamily.org/index.php?title=News:Eigen_3.3.4_released!`
>
> and extract it. Then in cmake_modules\FindEigen3.cmake set [...]

Thank you for the quick reply, when I specify the EIGEN3_INCLUDE_DIR to eigen/cmake, the cmake generates the error in below,

```
CMake Error at cmake_modules/FindEigen3.cmake:33 (file):
  file failed to open for reading (No such file or directory):

    E:/Tools/eigen-3.3.4/cmake/Eigen/src/Core/util/Macros.h
```
Could you please tell me how to solve this?

Zumbalamambo, 2020-04-05:
@shrddr Im getting "ImportError: DLL load failed: The specified module could not be found.". Any fix??

vgveerain, 2020-04-30:
Followed this through. I only get the private members for g2o after importing. It was recommended on a separate thread to build using sudo, but since I'm building on Windows, I'm stuck. Any help would be appriceated.

simon376, 2020-06-24:
i tried every cmake flag i could find trying to get it to use the 64bit compiler, but VS always uses MSBuild x86 :/ 

When i try this command

> "c:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\amd64\MSBuild.exe" g2o.sln /property:Configuration=Release

it fails because it cannot find the header files needed

can you help me out? :/ 
(I'm already using your fork btw, I didnt want to create another issue though)

LaurensJN, 2021-06-16:
Hi, thanks for the elaborate steps to build on windows. 

I'm following all steps (using VCPKG for cholmod), but I get stuck with an error while building. I made sure that CMAKE printed _Found CHOLMOD and its dependencies_, however the linear_solver_cholmod.h cannot find the file cholmod.h. I did specify the includes directory correctly (I think). I'm using Microsoft Visual Studio 16 2019. 

My CHOLMOD variables are:

- CHOLMOD_INCLUDES: C:/src/vcpkg/installed/x64-windows/include/suitesparse
- CHOLMOD_LIBRARIES: C:/src/vcpkg/installed/x64-windows/lib/libcholmod.lib
- CHOLMOD_METIS_LIBRARY: C:/src/vcpkg/installed/x64-windows/lib/metis.lib

While generating build files I do get some errors about not having installed Qt5, but nothing in relation to Cholmod. Does anybody have the same experience, or does anybody know what I should do?

LaurensJN, 2021-06-16:
I relatively fast found a way to work around the issue. After the generation of the build files I just replaced the `#include <cholmod.h>` that gave the error with `#include <EXTERNAL/suitesparse-metis-for-windows-1.5.0/build/install/include/suitesparse/cholmod.h>`, that is provided with the github repo. I don't think that is correct but it did work. 😄

Garrison-ctrl, 2024-05-06:
[image]