uoip/g2opy #31: How to set camera intrinsics for edge?

medakk, 2019-05-29:
For EdgeSE3ProjectXYZ, we can set the camera intrinsics with the public fx, fy, cx, cy values, in the [g2o codebase](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/sba/types_six_dof_expmap.h#L228).

How do I set these values in g2opy?

medakk, 2019-05-29:
For now, I changed the definition of EdgeSE3ProjectXYZ in [this file](https://github.com/uoip/g2opy/blob/5587024b17fd812c66d91740716fbf0bf5824fbc/python/types/sba/types_six_dof_expmap.h#L81):

```cpp
    // Projection using focal_length in x and y directions
    py::class_<EdgeSE3ProjectXYZ, BaseBinaryEdge<2, Vector2D, VertexSBAPointXYZ, VertexSE3Expmap>>(m, "EdgeSE3ProjectXYZ")
        .def(py::init<>())
        .def("compute_error", &EdgeSE3ProjectXYZ::computeError)
        .def("is_depth_positive", &EdgeSE3ProjectXYZ::isDepthPositive)
        .def("linearize_oplus", &EdgeSE3ProjectXYZ::linearizeOplus)
        .def("cam_project", &EdgeSE3ProjectXYZ::cam_project)
        .def_readwrite("fx", &EdgeSE3ProjectXYZ::fx)
        .def_readwrite("fy", &EdgeSE3ProjectXYZ::fy)
        .def_readwrite("cx", &EdgeSE3ProjectXYZ::cx)
        .def_readwrite("cy", &EdgeSE3ProjectXYZ::cy)
    ;
```

yiakwy-enterprise-roborock, 2020-04-20:
@medakk does it works well ? @uoip Some projects such as ORB-SLam use the same way to add poses as vertices.

medakk, 2020-04-20:
@yiakwy-enterprise-roborock Its been a while since I used g2opy, but IIRC it worked fine.

yiakwy-enterprise-roborock, 2020-04-21:
I confirmed that modifying these attributes works well with EdgeSE3ProjectXYZ python bindings. @medakk I suggested mark this question as resolved.

medakk, 2020-04-21:
I wouldn't consider it resolved yet: ideally this should be a part of the g2opy codebase, and I'm not sure how many other classes have such missing attributes.