uoip/g2opy #34: malloc(): memory corruption

a1qq2345609, 2019-09-03:
I've successfully installed g2o on Python 3.6.8 following the instructions.
But when I import g2o, the following error occurred:
`>>> import g2o
malloc(): memory corruption
Aborted (core dumped)`
and my python exits automatically.
I'm using Ubuntu 18.04.1 as a virtual machine on my computer. Is it an issue related to my system or just a bug in g2opy?

a1qq2345609, 2019-09-03:
What's more, now I switched to Ubuntu 16.04.3 and successfully installed g2o on Python 3.5.2. However, an Segmentation fault occurred when I import g2o:
`>>> import g2o`
`Segmentation fault (core dumped)`
and python exits automatically again.
When I installed g2o at the last step in the instructions, there's the information shown in the terminal:
`$ sudo python3.5 setup.py install`
`running install`
`copying ./lib/g2o.cpython-35m-x86_64-linux-gnu.so -> /usr/lib/python3/dist-packages`
Is there something about `python3` or `python3.5`? But there's no folder named `dist-packages` under `python3.5`...

lookingz, 2019-09-13:
Try install package "libqglviewer-dev-qt4", then rebuild g2o and g2opy.

a1qq2345609, 2019-09-18:
> Try install package "libqglviewer-dev-qt4", then rebuild g2o and g2opy.

Thanks a lot. I got the wrong way to rebuild g2opy...

closed 2019-09-18 by a1qq2345609