uoip/g2opy #35: Error in making g2opy File

ajinkyabedekar, 2019-11-12:
While running make -j8 command, I am getting the following error:

```
ajinkya@DESKTOP-HUJCKK6:~$ cd g2opy/build/
ajinkya@DESKTOP-HUJCKK6:~/g2opy/build$ make -j8
[  1%] Built target opengl_helper
[  6%] Built target freeglut_minimal
[  8%] Built target stuff
[ 34%] Built target csparse
[ 35%] Built target csparse_extension
[ 48%] Built target core
[ 49%] Built target solver_pcg
[ 50%] Built target solver_dense
[ 51%] Built target solver_csparse
[ 52%] Built target solver_structure_only
[ 53%] Built target solver_cholmod
[ 61%] Built target types_slam2d
[ 62%] Built target solver_eigen
[ 64%] Built target solver_slam2d_linear
[ 73%] Built target types_slam3d
[ 77%] Built target types_data
[ 80%] Built target types_sclam2d
[ 82%] Built target types_sba
[ 87%] Built target types_slam2d_addons
[ 93%] Built target types_slam3d_addons
[ 94%] Built target test_isometry3d_mappings
[ 96%] Built target test_mat2quat_jacobian
[ 96%] Built target test_slam3d_jacobian
[ 97%] Built target types_icp
[ 98%] Built target types_sim3
[ 99%] Built target contrib
[100%] Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
In file included from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:16:0,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/ajinkya/g2opy/python/g2o.cpp:1:
/home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/detail/internals.h:82:14: warning: 'int PyThread_create_key()' is deprecated [-Wdeprecated-declarations]
     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
              ^
In file included from /usr/include/python3.7m/pystate.h:11:0,
                 from /usr/include/python3.7m/traceback.h:8,
                 from /usr/include/python3.7m/Python.h:119,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/detail/common.h:111,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/pytypes.h:12,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:13,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/ajinkya/g2opy/python/g2o.cpp:1:
/usr/include/python3.7m/pythread.h:95:17: note: declared here
 PyAPI_FUNC(int) PyThread_create_key(void) Py_DEPRECATED(3.7);
                 ^
In file included from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:16:0,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/ajinkya/g2opy/python/g2o.cpp:1:
/home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/detail/internals.h:82:14: warning: 'int PyThread_create_key()' is deprecated [-Wdeprecated-declarations]
     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
              ^
In file included from /usr/include/python3.7m/pystate.h:11:0,
                 from /usr/include/python3.7m/traceback.h:8,
                 from /usr/include/python3.7m/Python.h:119,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/detail/common.h:111,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/pytypes.h:12,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:13,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/ajinkya/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/ajinkya/g2opy/python/g2o.cpp:1:
/usr/include/python3.7m/pythread.h:95:17: note: declared here
 PyAPI_FUNC(int) PyThread_create_key(void) Py_DEPRECATED(3.7);
                 ^
[... 499 lines cut ...]
   EIGEN_DEVICE_FUNC inline NonConstCoeffReturnType w() { return this->derived().coeffs().w(); }
                                                    ^
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:72:44: note:                 Eigen::QuaternionBase<Derived>::CoeffReturnType Eigen::QuaternionBase<Derived>::w() const [with Derived = Eigen::Quaternion<double>; Eigen::QuaternionBase<Derived>::CoeffReturnType = const double&]
   EIGEN_DEVICE_FUNC inline CoeffReturnType w() const { return this->derived().coeffs().coeff(3); }
                                            ^
In file included from /home/ajinkya/g2opy/python/g2o.cpp:19:0:
/home/ajinkya/g2opy/python/core/base_unary_edge.h: At global scope:
/home/ajinkya/g2opy/python/core/base_unary_edge.h:33:40: warning: unused parameter 'm' [-Wunused-parameter]
 void declareBaseUnaryEdge(py::module & m) {
                                        ^
In file included from /home/ajinkya/g2opy/python/g2o.cpp:20:0:
/home/ajinkya/g2opy/python/core/base_binary_edge.h:40:41: warning: unused parameter 'm' [-Wunused-parameter]
 void declareBaseBinaryEdge(py::module & m) {
                                         ^
In file included from /home/ajinkya/g2opy/python/g2o.cpp:25:0:
/home/ajinkya/g2opy/python/core/linear_solver.h:38:39: warning: unused parameter 'm' [-Wunused-parameter]
 void declareLinearSolver(py::module & m) {
                                       ^
make[2]: *** [python/CMakeFiles/g2o.dir/build.make:63: python/CMakeFiles/g2o.dir/g2o.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1520: python/CMakeFiles/g2o.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
ajinkya@DESKTOP-HUJCKK6:~/g2opy/build$
```

Can someone please help?

closed 2019-12-03 by ajinkyabedekar

rancheng, 2020-01-05:
are you compiling against `Eigen 3.3.X`?

rancheng, 2020-01-05:
in `Eigen 3.3.5` and above, the `Eigen::Quaternion` modified their interface after the update, eigen changed the return type of the function:

```cpp
Eigen::Quaternion<float> q = {1,0,0,0}; // w x y z
cout << q.w() << endl;
```

in `Quaternion.h` (`Eigen 3.3.7`):
```cpp
EIGEN_DEVICE_FUNC inline NonConstCoeffReturnType w() { return this->derived().coeffs().w(); }
```

in `Quaternion.h` (`Eigen 3.3.4`)
```cpp
EIGEN_DEVICE_FUNC inline Scalar& w() { return this->derived().coeffs().coeffRef(3); }
```

the return type is a `Scalar` type before.

Now define of `NonConstCoeffReturnType` is as following:
```cpp
typedef typename internal::conditional<bool(internal::traits<Derived>::Flags&LvalueBit),
                                        Scalar&, CoeffReturnType>::type NonConstCoeffReturnType;
```

This is a type cast issue, what we can do is to convert the `NonConstCoeffReturnType` to `Scalar&`, unlucky I don't know how to do that yet, if anyone is familiar with this, please post your solution.

BavanthaU, 2022-04-28:
This workaround fixed my problem, please check this one if you are facing this issue. 

https://github.com/pybind/pybind11/issues/2058#issuecomment-693329238

Change g2opy/python/core/eigen_types.h, Line 185-187 with following. 

Return type of Eigen::Quaterniond::x() const is const double &, and return type of Eigen::Quaterniond::x() is double &. So you can change the return type of these functions like this:
```
.def("x", (const double &(Eigen::Quaterniond::*)() const) & Eigen::Quaterniond::x)
.def("y", (const double &(Eigen::Quaterniond::*)() const) & Eigen::Quaterniond::y)
.def("z", (const double &(Eigen::Quaterniond::*)() const) & Eigen::Quaterniond::z)
.def("w", (const double &(Eigen::Quaterniond::*)() const) & Eigen::Quaterniond::w)
```