uoip/g2opy #38: AttributeError: module 'g2o' has no attribute 'SparseOptimizer'

kaixinbear, 2020-03-03:
Installing with sudo (I was okay with that since I was using it in a container) seemed to work.

Naturally that would be in the system python. I don't quite understand the issue though? Do some modules under the Cmake file have permission requirements?

_Originally posted by @varun19299 in https://github.com/uoip/g2opy/issues/15#issuecomment-450617698_

Successfully build,make,install g2o,but I can't use the module although making with sudo.
Could someone help me?

minhkhang1795, 2020-03-05:
As discussed [here](https://github.com/uoip/g2opy/issues/9), you need to build for the python version you're using. Remove the `build` folder and re-run cmake with:
```
cmake -DPYBIND11_PYTHON_VERSION=3.6 ..
```
Note: don't run any `export PYTHON_PATH=PATH-to-g2o/g2o:$PYTHONPATH` because the `g2o` folder is in C++ and not what you want.

chungtdnatv, 2021-01-27:
> As discussed [here](https://github.com/uoip/g2opy/issues/9), you need to build for the python version you're using. Remove the `build` folder and re-run cmake with:
> 
> ```
> cmake -DPYBIND11_PYTHON_VERSION=3.6 ..
> make -j8
> ```
> 
> Note: don't run any `export PYTHON_PATH=PATH-to-g2o/g2o:$PYTHONPATH` because the `g2o` folder is in C++ and not what you want.

Chao ban. Minh build lai roi ma van bi loi a. Ban chi giup minh voi a. Minh cam on.