uoip/g2opy #42: Cannot make on Ubuntu

generalizedeigenvector, 2020-05-05:
When doing cmake .. I get the following error:

> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Compiling on Unix
> -- Could NOT find CHOLMOD (missing:  CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES)
> -- A library with BLAS API not found. Please specify library location.
> -- A library with BLAS API not found. Please specify library location.
> -- LAPACK requires BLAS.
> -- Could NOT find CSPARSE (missing:  CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY)
> -- Building LGPL code as static library (affects license of the binary)
> -- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so
> -- Compiling with OpenGL support
> CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:27 (message):
>   The imported target "Qt5::Gui" references the file
> 
>      "/usr/lib/x86_64-linux-gnu/libGL.so"
> 
>   but this file does not exist.  Possible reasons include:
> 
>   * The file was deleted, renamed, or moved to another location.
> 
>   * An install or uninstall procedure did not complete successfully.
> 
>   * The installation package was faulty and contained
> 
>      "/usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake"
> 
>   but not all the files it references.
> 
> Call Stack (most recent call first):
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:50 (_qt5_Gui_check_file_exists)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:74 (_qt5gui_find_extra_libs)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:158 (include)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:99 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5OpenGL/Qt5OpenGLConfig.cmake:99 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:26 (find_package)
>   cmake_modules/FindQGLViewer.cmake:1 (FIND_PACKAGE)
>   CMakeLists.txt:148 (FIND_PACKAGE)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/maintenance/git/g2opy/build/CMakeFiles/CMakeOutput.log".

I have a file called "/usr/lib/x86_64-linux-gnu/lib**E**GL.so" - perhaps I just need to rename it? Also tried to apt-get Qt5 anew but this did not help.

juanvillacres, 2021-12-31:
> 
Did you find a solution?

charlescatta, 2023-05-17:
Installing suitesparse fixed this issue for me on Ubuntu 20.04:
`sudo apt-get install -y libsuitesparse-dev`

For Mac OS users:
`brew install suitesparse`