uoip/g2opy #43: g2o_viewer ERROR

Yannnnnnnnnnnn, 2020-05-11:
It seems that there are some mistakes in g2o_viewer. When I tried to load  sphere.g2o , it always returned with 0 vertices and 0 measurements.
[image]

Besides, it is okay if I use the original g2o_viewer to load sphere.g2o.

HouBiaoLiu, 2022-01-19:
I met the same problem, then how to solve it?

HouBiaoLiu, 2022-01-19:
How to solver it?

Yannnnnnnnnnnn, 2022-01-22:
Since no one will respond to this question, I will close it.

I think it is not a good idea to use g2o in python with GUI.

closed 2022-01-22 by Yannnnnnnnnnnn

wsp666, 2022-01-30:
I have the same problem. The original g2o_viewer means old version? I don't use  g2o in python with GUI.