uoip/g2opy #46: make in Ubuntu error,I went to 99 percent but failed

1761865308, 2020-07-18:
Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
........
make[2]: *** [python/CMakeFiles/g2o.dir/build.make:63：python/CMakeFiles/g2o.dir/g2o.cpp.o] error 1
make[1]: *** [CMakeFiles/Makefile2:1345：python/CMakeFiles/g2o.dir/all] error 2
make: *** [Makefile:130：all] error 2

1761865308, 2020-08-10:
> Same here

I successly install g2opy in ubuntu 18.0 and python 3.8,you can replace your eigen3 by legacy version(eigen 3.4.4)，good luck for you

blarkj, 2020-08-11:
@1761865308 , Thank you so much! I will try it right away!

ValeriiaIZH, 2020-08-25:
I still have the same error.
ubuntu 18.04, python 3.8, eigen 3.3.4-4.

koide3, 2020-10-06:
I encountered the same problem on Ubuntu 20.04. In my case, I could build the library by changing:

```cpp
        .def("x", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::x)
        .def("y", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::y)
        .def("z", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::z)
        .def("w", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::w)
```

in ```g2opy/python/core/eigen_types.h``` to:

```cpp
        .def("x", [](const Eigen::Quaterniond& q) { return q.x(); })
        .def("y", [](const Eigen::Quaterniond& q) { return q.y(); })
        .def("z", [](const Eigen::Quaterniond& q) { return q.z(); })
        .def("w", [](const Eigen::Quaterniond& q) { return q.w(); })
```

adeelakram03, 2021-06-01:
I have the same issue. 

> @1761865308 , Thank you so much! I will try it right away!

I have same issue. Did you find solution ?

Herusyahputra, 2021-06-10:
hello everyone..

I the finding this problem, I use ubuntu 20.04 and python 3.8. I have been many more try but I can't find a solution. can you help me? 

fatal error: Python.h: No such file or directory
  111 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [python/CMakeFiles/g2o.dir/build.make:63: python/CMakeFiles/g2o.dir/g2o.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1345: python/CMakeFiles/g2o.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

Ali007788, 2021-11-09:
I still have the same problem! 
ubuntu 20.4, python 3.8.10 and eigen 3.3.7
Has anyone an idea how to solve it?

Twenkid, 2021-11-28:
@Ali007788 

Try the koide3's solution - it has just worked for me.

Ali007788, 2021-11-28:
@Twenkid 
I already tried it, but the building went to 99% then I got this #58! 
any tips?

Twenkid, 2021-11-28:
Maybe out of memory error?
Try: make -j3  -B 
Or -j1 (number of threads, -B -rebuild, -d debug;)
Watch the RAM usage with htop.  

BTW, that combination "make -j8" is often given by default, but sometimes it is too much - for virtual machines etc. and there are no notes about that in the building instructions.

mjunsen123, 2024-02-16:
Follow koide3's solution and downgrade my python from 3.11 to 3.9 works for m

keeperlibofan, 2024-06-05:
> I encountered the same problem on Ubuntu 20.04. In my case, I could build the library by changing:
> 
> ```c++
>         .def("x", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::x)
>         .def("y", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::y)
>         .def("z", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::z)
>         .def("w", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::w)
> ```
> 
> in `g2opy/python/core/eigen_types.h` to:
> 
> ```c++
>         .def("x", [](const Eigen::Quaterniond& q) { return q.x(); })
>         .def("y", [](const Eigen::Quaterniond& q) { return q.y(); })
>         .def("z", [](const Eigen::Quaterniond& q) { return q.z(); })
>         .def("w", [](const Eigen::Quaterniond& q) { return q.w(); })
> ```

It is useful!!! And I solved this problem.

ysy200187, 2025-05-20:
> I encountered the same problem on Ubuntu 20.04. In my case, I could build the library by changing:
> 
>         .def("x", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::x)
>         .def("y", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::y)
>         .def("z", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::z)
>         .def("w", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::w)
> in to:`g2opy/python/core/eigen_types.h`
> 
>         .def("x", [](const Eigen::Quaterniond& q) { return q.x(); })
>         .def("y", [](const Eigen::Quaterniond& q) { return q.y(); })
>         .def("z", [](const Eigen::Quaterniond& q) { return q.z(); })
>         .def("w", [](const Eigen::Quaterniond& q) { return q.w(); })

Ubuntu22.04 useful