uoip/g2opy #49: Pose Optimization

aggelen, 2020-11-22:
I'm trying to implement a simple pose optimizer with 3 frames. I've calculated 3D points by triangulation. Then, I projected some of these points onto the 3rd frame. I have 3D pts. and projected points. I need to refine the camera pose at third frame. I've already looked through the examples but I couldn't figure out how to optimize it. 

- Added a vertex as VertexSE3Expmap with a predicted pose for 3rd frame. (Vertex ID 0)
- Added all 3d points as VertexSBAPointXYZ. (Vertex IDs 1, 2, ...)

- Added an edge for all projected points as EdgeProjectXYZ2UV wtih
  - set_vertex(0, vertex(i+1))   -> looping with i
  - set_vertex(1, vertex(0))
  - edge.set_parameter_id(0, 0)

I'm not sure if this is the right method. It doesn't work for my data. It is like the image below. Green: actual point, red: projected point with predicted pose, magenta: projected point with optimized pose. Can you suggest anything? Thanks!

[image]

shaoxh, 2021-02-13:
I still cannot use this package. when import an error shows up. #50
Do you have any idea how to tickle it?

LukasBommes, 2021-07-23:
Take a look here: https://github.com/raulmur/ORB_SLAM2/blob/master/src/Optimizer.cc
The method Optimizer::PoseOptimization performs pose optimization.