uoip/g2opy #55: Cannot make on Jetson Xavier

aimiwer, 2021-08-15:
Hi.
I got following errors when doing "make -j8".
My environment is 
- Jetson Xavier
- Ubuntu 20.04
- all requirements are installed 

Error: 
[  1%] Building CXX object g2o/stuff/CMakeFiles/stuff.dir/sparse_helper.cpp.o
[  1%] Building CXX object g2o/stuff/CMakeFiles/opengl_helper.dir/opengl_primitives.cpp.o
[  1%] Building CXX object EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_font.cpp.o
[  1%] Building CXX object g2o/stuff/CMakeFiles/stuff.dir/timeutil.cpp.o
[  2%] Building CXX object g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/csparse_helper.cpp.o
[  3%] Building CXX object EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_stroke_mono_roman.cpp.o
[  4%] Building CXX object EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_stroke_roman.cpp.o
[  5%] Building CXX object g2o/stuff/CMakeFiles/stuff.dir/command_args.cpp.o
c++: error: unrecognized command line option ‘-msse4.2’
c++: error: unrecognized command line option ‘-msse4.2’
g2o/stuff/CMakeFiles/stuff.dir/build.make:75: recipe for target 'g2o/stuff/CMakeFiles/stuff.dir/timeutil.cpp.o' failed
make[2]: *** [g2o/stuff/CMakeFiles/stuff.dir/timeutil.cpp.o] Error 1
make[2]: *** 未完了のジョブを待っています....
c++: error: unrecognized command line option ‘-msse4.2’
c++: error: unrecognized command line option ‘-msse4.2’
EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/build.make:75: recipe for target 'EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_font.cpp.o' failed
make[2]: *** [EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_font.cpp.o] Error 1
c++: error: unrecognized command line option ‘-msse4.2’
make[2]: *** 未完了のジョブを待っています....
c++: error: unrecognized command line option ‘-msse4.2’
EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/build.make:89: recipe for target 'EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_stroke_mono_roman.cpp.o' failed
make[2]: *** [EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_stroke_mono_roman.cpp.o] Error 1
g2o/stuff/CMakeFiles/stuff.dir/build.make:103: recipe for target 'g2o/stuff/CMakeFiles/stuff.dir/sparse_helper.cpp.o' failed
make[2]: *** [g2o/stuff/CMakeFiles/stuff.dir/sparse_helper.cpp.o] Error 1
g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/build.make:75: recipe for target 'g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/csparse_helper.cpp.o' failed
g2o/stuff/CMakeFiles/opengl_helper.dir/build.make:75: recipe for target 'g2o/stuff/CMakeFiles/opengl_helper.dir/opengl_primitives.cpp.o' failed
make[2]: *** [g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/csparse_helper.cpp.o] Error 1
make[2]: *** [g2o/stuff/CMakeFiles/opengl_helper.dir/opengl_primitives.cpp.o] Error 1
CMakeFiles/Makefile2:1089: recipe for target 'g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/all' failed
make[1]: *** [g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/all] Error 2
make[1]: *** 未完了のジョブを待っています....
CMakeFiles/Makefile2:553: recipe for target 'g2o/stuff/CMakeFiles/opengl_helper.dir/all' failed
make[1]: *** [g2o/stuff/CMakeFiles/opengl_helper.dir/all] Error 2
c++: error: unrecognized command line option ‘-msse4.2’
c++: error: unrecognized command line option ‘-msse4.2’
EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/build.make:103: recipe for target 'EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_stroke_roman.cpp.o' failed
make[2]: *** [EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/freeglut_stroke_roman.cpp.o] Error 1
g2o/stuff/CMakeFiles/stuff.dir/build.make:89: recipe for target 'g2o/stuff/CMakeFiles/stuff.dir/command_args.cpp.o' failed
make[2]: *** [g2o/stuff/CMakeFiles/stuff.dir/command_args.cpp.o] Error 1
CMakeFiles/Makefile2:527: recipe for target 'EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/all' failed
make[1]: *** [EXTERNAL/freeglut/CMakeFiles/freeglut_minimal.dir/all] Error 2
CMakeFiles/Makefile2:579: recipe for target 'g2o/stuff/CMakeFiles/stuff.dir/all' failed
make[1]: *** [g2o/stuff/CMakeFiles/stuff.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

aimiwer, 2021-08-15:
I solved this problem.
I remove "-msse4.2" in CMakeLists.txt.

closed 2021-08-15 by aimiwer

beautifulchoi, 2023-11-18:
@aimiwer hi, I'm also running nvidia xavier nx, but I failled while in 99% process. did you met the problem or solve it?