uoip/g2opy #58: Error in installing on ubuntu 20.04

Ali007788, 2021-11-09:
I'm trying to install the library on **ubuntu 20.04, python 3.8.10 and eigen 3.3.7.** 
I already tried this solution #48 but while running make -j8 command, I got this error:

`[ 99%] Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
In file included from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:16,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/internals.h:82:34: warning: ‘int PyThread_create_key()’ is deprecated [-Wdeprecated-declarations]
   82 |     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
      |                                  ^
In file included from /usr/include/python3.8/pystate.h:10,
                 from /usr/include/python3.8/genobject.h:11,
                 from /usr/include/python3.8/Python.h:121,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/common.h:111,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pytypes.h:12,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/usr/include/python3.8/pythread.h:100:36: note: declared here
  100 | Py_DEPRECATED(3.7) PyAPI_FUNC(int) PyThread_create_key(void);
      |                                    ^~~~~~~~~~~~~~~~~~~
In file included from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:16,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/internals.h:82:34: warning: ‘int PyThread_create_key()’ is deprecated [-Wdeprecated-declarations]
   82 |     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
      |                                  ^
In file included from /usr/include/python3.8/pystate.h:10,
                 from /usr/include/python3.8/genobject.h:11,
                 from /usr/include/python3.8/Python.h:121,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/common.h:111,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pytypes.h:12,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/usr/include/python3.8/pythread.h:100:36: note: declared here
  100 | Py_DEPRECATED(3.7) PyAPI_FUNC(int) PyThread_create_key(void);
      |                                    ^~~~~~~~~~~~~~~~~~~
In file included from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:16,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/internals.h:82:34: warning: ‘int PyThread_create_key()’ is deprecated [-Wdeprecated-declarations]
   82 |     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
      |                                  ^
In file included from /usr/include/python3.8/pystate.h:10,
                 from /usr/include/python3.8/genobject.h:11,
                 from /usr/include/python3.8/Python.h:121,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/common.h:111,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pytypes.h:12,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/usr/include/python3.8/pythread.h:100:36: note: declared here
  100 | Py_DEPRECATED(3.7) PyAPI_FUNC(int) PyThread_create_key(void);
      |                                    ^~~~~~~~~~~~~~~~~~~
In file included from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:16,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/attr.h:13,
                 from /home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:43,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:1:
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/detail/internals.h:82:34: warning: ‘int PyThread_create_key()’ is deprecated [-Wdeprecated-declarations]
   82 |     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
      |                                  ^
In file included from /usr/include/python3.8/pystate.h:10,
                 from /usr/include/python3.8/genobject.h:11,
[... 275 lines cut ...]
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:1852:9:   required from ‘bool pybind11::detail::argument_loader<Args>::load_impl_sequence(pybind11::detail::function_call&, pybind11::detail::index_sequence<Is ...>) [with long unsigned int ...Is = {0, 1}; Args = {Eigen::Matrix<double, 3, 1, 0, 3, 1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&}]’
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/cast.h:1832:50:   required from ‘bool pybind11::detail::argument_loader<Args>::load_args(pybind11::detail::function_call&) [with Args = {Eigen::Matrix<double, 3, 1, 0, 3, 1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&}]’
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:135:17:   required from ‘void pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = g2o::declareEigenTypes(pybind11::module&)::<lambda(Eigen::Matrix<double, 3, 1>&, Eigen::Matrix<double, 3, 1>&)>; Return = Eigen::Quaternion<double>; Args = {Eigen::Matrix<double, 3, 1, 0, 3, 1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&}; Extra = {pybind11::name, pybind11::scope, pybind11::sibling}]’
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:65:9:   required from ‘pybind11::cpp_function::cpp_function(Func&&, const Extra& ...) [with Func = g2o::declareEigenTypes(pybind11::module&)::<lambda(Eigen::Matrix<double, 3, 1>&, Eigen::Matrix<double, 3, 1>&)>; Extra = {pybind11::name, pybind11::scope, pybind11::sibling}; <template-parameter-1-3> = void]’
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:1097:22:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def_static(const char*, Func&&, const Extra& ...) [with Func = g2o::declareEigenTypes(pybind11::module&)::<lambda(Eigen::Matrix<double, 3, 1>&, Eigen::Matrix<double, 3, 1>&)>; Extra = {}; type_ = Eigen::Quaternion<double>; options = {}]’
/home/tufreiberg/g2opy/python/core/eigen_types.h:183:14:   required from here
/home/tufreiberg/g2opy/EXTERNAL/pybind11/include/pybind11/eigen.h:83:20: warning: implicitly-declared ‘Eigen::Stride<-1, -1>& Eigen::Stride<-1, -1>::operator=(const Eigen::Stride<-1, -1>&)’ is deprecated [-Wdeprecated-copy]
   83 |             stride = {EigenRowMajor ? rstride : cstride /* outer stride */,
      |             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   84 |                       EigenRowMajor ? cstride : rstride /* inner stride */ };
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:468,
                 from /home/tufreiberg/g2opy/g2o/core/eigen_types.h:30,
                 from /home/tufreiberg/g2opy/build/g2o/config.h:26,
[... cut ...]
                 from /home/tufreiberg/g2opy/python/core/hyper_graph.h:5,
                 from /home/tufreiberg/g2opy/python/g2o.cpp:3:
/usr/include/eigen3/Eigen/src/Core/Stride.h:71:5: note: because ‘Eigen::Stride<-1, -1>’ has user-provided ‘Eigen::Stride<InnerStrideAtCompileTime, OuterStrideAtCompileTime>::Stride(const Eigen::Stride<InnerStrideAtCompileTime, OuterStrideAtCompileTime>&) [with int _OuterStrideAtCompileTime = -1; int _InnerStrideAtCompileTime = -1]’
   71 |     Stride(const Stride& other)
      |     ^~~~~~

`

Has anyone an idea how to solve it?

Polydynamical, 2021-11-15:
I had a similar problem. I just deleted the g2opy folder, cloned this repo and just copy pasted the build instructions. After an hour of working on it, it magically works

Ali007788, 2021-11-15:
@Polydynamical 
Thanks for your reply. I'm confused and I think that I couldn't get what you mean by ' I just deleted the g2opy folder, cloned this repo and just copy pasted the build instructions' 
Which repo did you clone?
Can you explain the steps that you did to get it working? 
Thanks

Polydynamical, 2021-11-16:
I deleted everything related to g2opy

Ali007788, 2021-11-23:
@Polydynamical  I already tried what you said. But I still have the same problem! 
Can you tell me which version of eigen and pybind11 did you use by installation? 
Thanks

Twenkid, 2021-11-28:
Change make -j8 to make -j3 or -j2 or if still doesn't complete: -j1. Watch the memory usage during the build with htop.
If you add -B  - it will rebuild from scratch (clean). 

Given the point of break, I guess that may be out of memory error - at least I had similar issues, when building on a virtual machine with limited memory and with swap turned off. It seemed to need nearly 7 GB even for 3 threads.

 -j8 means 8 threads. So try with -j3 or -j2 etc. If you have a few cores and small memory some complicated parts of the compilation run out of resources.