uoip/g2opy #65: I met this problem when run python setup.py install

s8088899, 2022-04-18:
error: Multiple top-level packages discovered in a flat-layout: ['g2o', 'lib', 'script', 'contrib', 'EXTERNAL', 'cmake_modules'].

To avoid accidental inclusion of unwanted files or directories,
setuptools will not proceed with this build.

If you are trying to create a single distribution with multiple packages
on purpose, you should not rely on automatic discovery.
Instead, consider the following options:

1. set up custom discovery (`find` directive with `include` or `exclude`)
2. use a `src-layout`
3. explicitly set `py_modules` or `packages` with a list of names

To find more information, look for "package discovery" on setuptools docs.

Tomzxcv, 2022-04-19:
It's work for me: pip3 install 'setuptools<61'

RTae, 2022-04-28:
@s8088899 I found the solution !! use this script below instead
```
sudo python3 setup.py install
```
However, my recommendation you should copy the file name "g2o.cpython-{python-version}-{arch}.so" in lib folder into 
<path-to-python-lib>/lib/python{python-version-which-you-use}/site-package

MarceloContreras, 2022-08-31:
Two solutions have gave me good results:

1. Just as @RTae said, you can move manually "g2o.so" to your python package folder. Be careful if you are using a conda enviorement, because then the env itself has its package folder.
2. Downgrade installed setuptool version to 58.2.0 with this command:

`pip install setuptools==58.2.0`

NaleRaphael, 2023-02-17:
I rewrote the file "setup.py" to make it work with `pip` or other package managers automatically. Hope this help.

You can simply replace the existing "setup.py" with the following script, and run `pip install ./` under the root directory of "g2opy".

```python
from setuptools import setup, find_packages
from pathlib import Path

THIS_DIR = Path(__file__).parent.absolute()

DIR_LIB = THIS_DIR.joinpath('./lib').absolute()
LIB_FILES = list(DIR_LIB.rglob('g2o*'))
PY_INIT = DIR_LIB.joinpath('__init__.py')

__version__ = '0.0.1'

def init_py_package():
    if not DIR_LIB.exists():
        raise RuntimeError('Directory of g2o libraries does not exist at {}'.format(DIR_LIB))
    if len(LIB_FILES) == 0:
        raise RuntimeError('No desired library files exist')

    # Create "__init__.py" inside "lib" directory to make it searchable by `find_package()`.
    # The remaining things can be handled by "pip".
    if not PY_INIT.exists():
        PY_INIT.touch()

def setup_package():
    excluded = []
    package_data = {}

    init_py_package()

    metadata = dict(
        name='g2opy',
        version=__version__,
        description='Python binding of C++ graph optimization framework g2o.',
        url='https://github.com/uoip/g2opy',
        license='BSD',
        packages=find_packages(where=DIR_LIB, exclude=excluded),
        keywords='g2o, SLAM, BA, ICP, optimization, python, binding',
        long_description="""This is a Python binding for c++ library g2o
            (https://github.com/RainerKuemmerle/g2o).
            g2o is an open-source C++ framework for optimizing graph-based nonlinear
            error functions. g2o has been designed to be easily extensible to a wide
            range of problems and a new problem typically can be specified in a few
            lines of code. The current implementation provides solutions to several
            variants of SLAM and BA."""
    )

    setup(**metadata)

if __name__ == '__main__':
    setup_package()
```

s8088899, 2023-02-18:
Thanks a lot！I already solve this problem by improving the network environment and virtual machine  versions.

robofar, 2024-05-02:
Package is installed, but when I run `conda list | grep g2o` why it cannot "see" this package?