uoip/g2opy #66: Photometric BA

artykov1511, 2022-07-13:
Hi. Thanks for wonderful work. I wonder which class do we need to inherit from in order to create custom edge class for photometric BA? Or how to implement photometric BA in general? Thanks in advance!

1280032578, 2022-07-13:
Excuse me
Hello, how do you install the g2opy library?
I'm running make, and 99% of it appears

In file included from /home/user/Desktop/envirment/g2opy/python/g2o. cpp:19:0:
/home/user/Desktop/envirment/g2opy/python/core/base_ unary_ edge. h: At global scope:
/home/user/Desktop/envirment/g2opy/python/core/base_ unary_ edge. h:33:6: warning: unused parameter ‘m’ [-Wunused-parameter]
void declareBaseUnaryEdge(py::module & m) {
In file included from /home/user/Desktop/envirment/g2opy/python/g2o. cpp:20:0:
/home/user/Desktop/envirment/g2opy/python/core/base_ binary_ edge. h:40:6: warning: unused parameter ‘m’ [-Wunused-parameter]
void declareBaseBinaryEdge(py::module & m) {
In file included from /home/user/Desktop/envirment/g2opy/python/g2o. cpp:25:0:
/home/user/Desktop/envirment/g2opy/python/core/linear_ solver. h:38:6: warning: unused parameter ‘m’ [-Wunused-parameter]
void declareLinearSolver(py::module & m) {

I don't know how to deal with it. I wonder if you can provide some valuable installation experience?
Say thank you in advance!!！

cheukwaylee, 2022-09-09:
> Hi. Thanks for wonderful work. I wonder which class do we need to inherit from in order to create custom edge class for photometric BA? Or how to implement photometric BA in general? Thanks in advance!

I am trying to develop a VO demo using direct method in Python, it should need g2o as its optimizer. is it the photometric BA?

cheukwaylee, 2022-09-09:
@artykov1511 
I am trying to develop a VO demo using direct method in Python, it should need g2o as its optimizer. is it the photometric BA?