uoip/g2opy #67: make[99%] error &&  void declareXXXX(py::module & m) {^}

1280032578, 2022-07-13:
I'm running make, and 99% of it appears

In file included from /home/user/Desktop/envirment/g2opy/python/g2o. cpp:19:0:
/home/user/Desktop/envirment/g2opy/python/core/base_ unary_ edge. h: At global scope:
/home/user/Desktop/envirment/g2opy/python/core/base_ unary_ edge. h:33:6: warning: unused parameter ‘m’ [-Wunused-parameter]
void declareBaseUnaryEdge(py::module & m) {
In file included from /home/user/Desktop/envirment/g2opy/python/g2o. cpp:20:0:
/home/user/Desktop/envirment/g2opy/python/core/base_ binary_ edge. h:40:6: warning: unused parameter ‘m’ [-Wunused-parameter]
void declareBaseBinaryEdge(py::module & m) {
In file included from /home/user/Desktop/envirment/g2opy/python/g2o. cpp:25:0:
/home/user/Desktop/envirment/g2opy/python/core/linear_ solver. h:38:6: warning: unused parameter ‘m’ [-Wunused-parameter]
void declareLinearSolver(py::module & m) {

I don't know how to deal with it. I wonder if you can provide some valuable installation experience?
Say thank you in advance!!！

Weiyi-Zhang258, 2022-07-16:
#48 may help

1280032578, 2022-10-05:
> #48 may help

Thanks, I have solved it. It is caused by insufficient memory of your computer