uoip/g2opy #82: Error while making

jeremy7703, 2024-10-25:
In file included from /home/jeremy/vis_nav_player/g2opy/python/g2o.cpp:1:
/home/jeremy/vis_nav_player/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h: At global scope:
/home/jeremy/vis_nav_player/g2opy/EXTERNAL/pybind11/include/pybind11/pybind11.h:1088:13: error: ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const char*, Func&&, const Extra& ...) [with Func = pybind11::detail::initimpl::constructor<Args>::execute(Class&, const Extra& ...) [with Class = pybind11::class_<Eigen::Rotation2D<double> >; Extra = {}; typename std::enable_if<(! Class::has_alias), int>::type <anonymous> = 0; Args = {const Eigen::Matrix<double, 2, 2, 0, 2, 2>&}]::<lambda(pybind11::detail::value_and_holder&, const Eigen::Matrix<double, 2, 2>&)>; Extra = {pybind11::detail::is_new_style_constructor}; type_ = Eigen::Rotation2D<double>; options = {}]’, declared using local type ‘pybind11::detail::initimpl::constructor<Args>::execute(Class&, const Extra& ...) [with Class = pybind11::class_<Eigen::Rotation2D<double> >; Extra = {}; typename std::enable_if<(! Class::has_alias), int>::type <anonymous> = 0; Args = {const Eigen::Matrix<double, 2, 2, 0, 2, 2>&}]::<lambda(pybind11::detail::value_and_holder&, const Eigen::Matrix<double, 2, 2>&)>’, is used but never defined [-fpermissive]
 1088 |     class_ &def(const char *name_, Func&& f, const Extra&... extra) {
      |             ^~~
make[2]: *** [python/CMakeFiles/g2o.dir/build.make:63: python/CMakeFiles/g2o.dir/g2o.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1345: python/CMakeFiles/g2o.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

ptrem, 2024-11-14:
I get the same error. Any solution on that?

adeelferozmirza, 2024-11-18:
use this, this repo is no more useful

https://github.com/RainerKuemmerle/g2o

jeremy7703, 2024-11-23:
> use this, this repo is no more useful
> 
> https://github.com/RainerKuemmerle/g2o

I am afraid that link is g2o written in C++, does that work well with my project written in python?

ptrem, 2024-11-23:
As described in: https://github.com/miquelmassot/g2o-python/issues/40

I used this repo: https://github.com/miquelmassot/g2o-python
and the pymem branch from https://github.com/RainerKuemmerle/g2o.git.
Worked for me. I also use it inside my Python Project.

shakurShirajul, 2025-04-21:
@ptrem i follow the step you given but it is giving this error in windows10

```
Python 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import g2o
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Shakur\AppData\Roaming\Python\Python311\site-packages\g2o\__init__.py", line 1, in <module>
    from .g2opy import *  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing g2opy: The specified module could not be found.
>>>
```